SBG-Systems/sbg_ros_driver #1: Z-axis acceleration in spite of being still

martin-velas, 2017-04-03:
Hi!
First of all - thanks for your great work! I'm trying to use your driver and it looks to be working (according to the `rostopic echo` and `rviz` visualization) except the acceleration in Z-axis which is constantly around -9.7 for sensor **lying still** on my desk. This is the output of ros echo:
```
header: 
  seq: 13202
  stamp: 
    secs: 1491219289
    nsecs: 732971334
  frame_id: map
orientation: 
  x: -0.000194159161765
  y: -0.00108367542271
  z: -0.967581689358
  w: -0.25255587697
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00231326837093
  y: 0.000343044259353
  z: -0.00297229643911
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0124550322071
  y: -0.0125222699717
  z: -9.78081607819
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

When I move with IMU, all other values changes accordingly just fine.

Do you have any tips for fixing this?
Thanks, Matin.

martin-velas, 2017-04-03:
It is probably caused by gravity - is it any way to cancel it by your driver?

ThomasLeMezo, 2017-04-04:
Hi Martin,

This is indeed due to gravity. It is difficult to "cancel" gravity as it depends on the place (mainly altitude but not only) where you are ([https://en.wikipedia.org/wiki/Gravity_of_Earth](https://en.wikipedia.org/wiki/Gravity_of_Earth)). It would be difficult to define a constant to add for every user.

I would recommend:
- Add a node to "cancel" earth gravity
- OR edit line 57 of the file and add a constant:
`imu_msg.linear_acceleration.z = pLogData->imuData.accelerometers[2] - earth_gravity;`

I will look at how to properly add this feature to the code next week.

closed 2018-03-11 by ThomasLeMezo