SBG-Systems/sbg_ros_driver #4: sbgEComHandle Error SBG_NOT_READY

Rumpel-StiltSkin, 2017-10-13:
Hi Guys, I currently have an SBG Eclipse that I am trying to integrate. 
I managed to chance upon this driver which is great.
However, when i run the driver, it keeps printing `[ERROR] [1507812084.649448000]: sbgEComHandle ErrorSBG_NOT_READY: A device isn't ready (Rx isn't ready for example).`

I have traced it to `sbg_ros_driver/sbgECom/src/sbgECom.c` which is called the `do while` loop to read all the frames. The comments above the function states that it is supposed to receive a `SBG_NO_ERROR`. But I am not sure why I am getting the `SBG_NOT_READY`. Could anybody verify whether this is normal?

swarmt, 2017-10-28:
I just connected my Ellipse-2D for the first time yesterday and did not see this error. Does the model that you are using have an initialization delay?

AtDinesh, 2017-12-19:
Hi !

I am facing the same error. But in addition, I also have the following one before : 

> sbgEComCmdGetInfo Error: SBG_TIME_OUT: We have a time out during frame reception.

Any idea of how to fix this ?
Thanks

ThomasLeMezo, 2018-03-11:
A new version of the driver has been released (under beta test for now). It should correct the errors you had.

closed 2018-03-11 by ThomasLeMezo