SBG-Systems/sbg_ros_driver #5: Couldn't find node when building

Ignat-Georgiev, 2018-03-10:
Whenever I try to build the driver myself, ROS is unable to find the node and launch it.

If I try to start it with the launch file it simply states that the `ellipse` node can't be found. If I try to launch the node manually I still can't find it.
*Note: I've been sourcing after building*

I managed to get over this issue simply by adding `catkin_package()` in the CmakeList.

Is this expected behaviour or should I submit a pull request?

ThomasLeMezo, 2018-03-10:
No: it is not the expected behavior. Did you checkout to the 1.0.7 version (the last version on the repo doesn't work) ?
I'm currently restarting the whole node (see devel branch): a stable version should be release this week.

Ignat-Georgiev, 2018-03-11:
I used the master branch (seems to be v1.0.6).

Just tested v1.0.7 and it doesn't have this issue. Is the new stable version you spoke of meant to have any major changes?

ThomasLeMezo, 2018-03-11:
Yes there will be major changes: it will better implement the SBG library. New messages and topics as been created. It will be possible to configure the IMU with a yaml file and to proceed to a magnetic calibration throw a ROS service.

closed 2018-03-11 by ThomasLeMezo