SBG-Systems/sbg_ros_driver #8: Non ASCII characters in message definitions are problematic

nicolaje, 2018-06-26:
Hello again,

non-ASCII characters in msg definitions are problematic even in comments :
https://github.com/ros/ros_comm/issues/143

This is especially an issue when replaying rosbags and trying to apply filters : this causes the filters to fail.

Characters such as "µ, °, σ" should be removed from the message definitions, e.g:
https://github.com/ENSTABretagneRobotics/sbg_ros_driver/blob/f56ede11bcf29f3456e50a4565ac6baaff6d2d11/msg/SbgEkfEuler.msg#L5

Thank you for this driver!

Jeremy

ThomasLeMezo, 2018-07-19:
The message file has been changed

nicolaje, 2018-10-31:
Hi @ThomasLeMezo , that is still an issue with the other messages being published. Example:

```bash
rosbag filter 2018-10-26-11-19-42.bag 2018-10-26-11-19-42_1.bag "t.secs <= 1540545740.895"
 2018-10-26-11-19-42_1.bag                                      0%   0.0 KB /  28.5 GB --:-- ETATraceback (most recent call last):
  File "/opt/ros/melodic/bin/rosbag", line 35, in <module>
    rosbag.rosbagmain()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/rosbag_main.py", line 1012, in rosbagmain
    cmds[cmd](argv[2:])
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/rosbag_main.py", line 372, in filter_cmd
    for topic, raw_msg, t, conn_header in inbag.read_messages(raw=True, return_connection_header=True):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 2679, in read_messages
    yield self.seek_and_read_message_data_record((entry.chunk_pos, entry.offset), raw, return_connection_header)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 2842, in seek_and_read_message_data_record
    msg_type = _get_message_type(connection_info)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 1895, in _get_message_type
    message_type = genpy.dynamic.generate_dynamic(info.datatype, info.msg_def)[info.datatype]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/dynamic.py", line 168, in generate_dynamic
    tmp_file.file.write(full_text.encode())
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 513: ordinal not in range(128)
```

here the special characters in the other sbg messages prevent the use of the filter command

jmount1992, 2019-01-25:
Hi @ThomasLeMezo  there are still special characters in some of the message definitions. You can easily check a file's encoding by going to the sbg_driver/msg folde rand running

`file *.msg`

It will tell you which have UTF-8 encoding. I know the following files had special characters:

- SbgEkfNav
- SbgEkfQuat
- SbgEvent
- SbgGpsHdt
- SbgGpsPos
- SbgGpsVel
- SbgImuData
- SbgOdoVel
- SbgPressure
- SbgShipMotion
- SbgUtcTime

closed 2019-01-25 by ThomasLeMezo