SBG-Systems/sbg_ros_driver #11: SBG_Ellipse not connecting

fgrigis, 2018-10-29:
Hi

We're using your ros wrapper for the sbg_ellipse 2 micro. We use the IMU in an UAV application where we've used the wrapper on a FitPC with an RS-232 interface, which worked fine for us. We had to chance our onvoard computer to an alternative which doesn't provide an RS-232 interface. Therefore we're connecting via a USB-to-RS232 dongle to the IMU. The node is starting up and running, all topics are advertised but there's no data on these topics. During the start up process we get two SBG_TIME_OUT warnings printed on the console, as you can see in the attached screenshot.

[image]

Is there any issue using the ros wrapper with a USB device (device the node should open is set correctly, in our case "/dev/ttyUSB0")? Thank you for your help.

ThomasLeMezo, 2018-10-29:
Our ros wrapper is based on an old SDK version therefore it might not be compatible with the new ellipse 2 micro. However, what's the baudrate of your link and did you test your imu with the sbg software?

fgrigis, 2018-10-30:
It was a trivial mistake on hardware side.. had to change TX & RX lines and it worked fine.

closed 2018-10-30 by fgrigis