SBG-Systems/sbg_ros_driver #12: SBG_TIME_OUT

louispetitjeanforssea, 2018-11-15:
Hi, 

I'm trying to use your driver on a Jetson TX2, plugging an SBG ellipse micro 2 in RS422. 
Will your driver be abble to deal directly with it, or do I need a RS422/USB converter ? 

Best regards, 
Louis

ThomasLeMezo, 2018-11-15:
I have no idea, but I guess it might work directly if you set the "portName" and "baudRate" correctly in the config file, see here: 
https://github.com/ENSTABretagneRobotics/sbg_ros_driver/blob/7ea946bde87e7a6712377acf15a06139a3627c3b/config/ellipse_A_default.yaml#L9
Let me know if it works,
Thomas

louispetitjeanforssea, 2018-11-15:
Thanks for your quick answer, 

I did set the baudrate and the port name but nothing happens, I keep getting this time out.
I tried all the serials port of the Jetson to be sure, and I set the Baudrate with the windows API. 
I'm really stuck now :/

Louis

ThomasLeMezo, 2018-11-15:
What do you mean by "Windows API"? 
* Have you configured the Ellipse with SbgCenter and set correctly the port & baudrate? 
* Is the port ID the same as with RS232 (PORT_A)?
* Have you tested the serial link with an other device to be sure it works (for instance two Jetson connected throw RS422)?

louispetitjeanforssea, 2018-11-16:
Hi, sorry for my late answer, I've had a problem with my SBG.

By windows API, I mean SBG center. 

- I think it's well configurated, I did set the bodrate and the port. 

[image]

- Yes the serial link has been tested

ThomasLeMezo, 2018-11-16:
You could try the "minimal ellipse" program included in the SDK from SBG to know if it is an issue with the ROS wrapper or the SDK.

louispetitjeanforssea, 2018-11-19:
Yes, I will test it today. Thanks for your help !

ThomasLeMezo, 2018-11-19:
> Hi again,
> After some tests, I know manage to access the datas of the central with the minimal ellipse program. I think that it is a ROS wrapper problem...

Could you send the code you used?

Thomas

louispetitjeanforssea, 2018-11-19:
Just the code of the sdk actually, the ellipseMinimal.c, compiled for ubuntu with gcc

closed 2019-05-23 by louispetitjeanforssea