SBG-Systems/sbg_ros_driver #21: Add udev rules to the documentation

ThomasLeMezo, 2019-05-16:
For Ellipse (to be check) :
- Add a new file `99-sbg.rules` in `/etc/udev/rules.d/`
`SUBSYSTEM=="tty", ATTRS{serial}=="FT0CQ26X", ATTRS{product}=="USB-RS232 Cable", SYMLINK+="sbg", GROUP="dialout", MODE="0660"`
- This allows to map the `/dev/ttyUSBx` to `/dev/sbg` which avoid to modify the port in configuration if it has changed. The default configuration file could be changed accordingly.

ErwanCL, 2019-06-03:
Udev rules have been added to documentation, with an example (screenshot and corresponding rules file).

The configuration file with the port `/dev/sbg` defined by udev rules, has been tested and validated on Ubuntu 18.04

ThomasLeMezo, 2019-06-03:
Is there different version for the "ATTRS{serial}"? It might depend on the version of the device (or the FTDI used)?

ErwanCL, 2019-06-03:
The serial attribute depends of the FTDI used, so the user, who wants to define the udev rules, has to check by himself the specific attributes (`udevadm info ..`) for its device.

The serial attributes does not depend on the sbg device, I have tested my USB-RS232 converter, defined by my udev rule, with two differents Ellipse, and it is still working.

closed 2019-06-03 by ThomasLeMezo