SBG-Systems/sbg_ros_driver #22: Problem of access to the data of /ekf_nav

mohand150, 2019-06-17:
Hello,
currently I work with an Ellipse2-A, I use sbg_driver on ROS to retrieve the data
in my project, I need 3 data:
[roll, pitch, yaw] (in rad) -> found on the topic **/ekf_euleur**
velocity (in m / s ^ 2) -> found on the topic **imu_data/accel** & **imu_data/gero**
speed (in m / s) -> **/mag**
I have a question, I found that according to the name of the topic **/ekf_euleur**, this data comes out of the kelman filter
but the topics**imu_data/accel** & **imu_data/gero** & **/mag** are there the output of the kelman filter? or are the raw data of the sensors?
Does the Ellipse2-A model have velocity data coming out of the EKF? the topic **/ekf_nav** does not receive any data
if yes, how to get them?
thanks for any help :)

ErwanCL, 2019-06-18:
Hello,

> the topicsimu_data/accel & imu_data/gero & /mag are there the output of the kelman filter? or are the raw data of the sensors?

These data are directly the data measured by the sensors.

> the topic /ekf_nav does not receive any data, if yes, how to get them?

You can receive data from the Ellispe. You just have to change the configuration file that you're using, and set the corresponding parameters. By default, it is set to 0 (disabled), you can change it according to the frequency output you want (Frequency is defined in the config file).

`# Position and velocities in NED coordinates with the accuracies on each axis`
` log_ekf_nav: 0`

mohand150, 2019-06-18:
thank you for your reply

> By default, it is set to 0 (disabled), you can change it according to the frequency output you want (Frequency is defined in the config file).

for the parameter **log_ekf_nav** I already modify it in the parameters to receive the data at **25 Hz** before posting this problem but I do not receive any data on this topic 

**Q1**-  **Does the ellipseA model generate postion and velocities via the EKF?** by subscribing to this topic ... I receive no data
1) I need these 3 data:
   1- **angular position** - vector3- (in radian)
   2- **linear speed** - vector3- (in m / s)
   3- **linear acceleration** - vector3- (in m / s ^ 2)
   4- **angular velocity** - vector3- (rad / s)

2) angular position I get it on the topic **/ekf_euleur**,I always need the **linear speed (m/s)**, **linear acceleration (m/s2)** and **angular velocity (rad/s)**.

**Q2**- **Is there a topic that publishes these data coming out of EKF?**

**Q3**- is the data of: imu_data / accel (m / s ^ 2) and imu_data / gero (rad / s)
are enough accurate for use on a drone or it will be necessary to pass  them in a kelman filter before?

cordially

ErwanCL, 2019-06-18:
> Q1- Does the ellipseA model generate postion and velocities via the EKF? by subscribing to this topic ... I receive no data

The EllipseA does NOT generate position and velocities via EKF, navigation is not enabled on this product.
https://www.sbg-systems.com/wp-content/uploads/Ellipse_Series_Leaflet.pdf

> Q2- Is there a topic that publishes these data coming out of EKF?

There is one, `ekf_nav` topic, but you won't be able to use it with your EllipseA.

> Q3- is the data of: imu_data / accel (m / s ^ 2) and imu_data / gero (rad / s)
are enough accurate for use on a drone or it will be necessary to pass them in a kelman filter before?

It depends on which data you need and how you use it. But sensors data at high frequency are pretty accurate, so you can use it for data coupling with other sources (like GPS) if you want to have accurate positioning.

closed 2019-06-18 by mohand150