SBG-Systems/sbg_ros_driver #26: Increasing ELLIPSE2-N ekf_euler output frequency to 1khz

Amir-Ramezani, 2019-11-14:
Hi,

I am using your ROS package to receive data from ELLIPSE2-N.
Currently I am able to receive **ekf_euler** at about 103hz after changing the config file. I actually don't need the GPS or BARO fusion, just need the Euler angles.

- Is it possible to increase the frquency to 1khz?
- At this moment, even though I set the frequency to be 200hz, but it is about 103 frames per second, is it because of serial baud rate? (how can I change this?)

Thanks in advance.

ErwanCL, 2019-11-25:
Hey,

Sorry for late reply.
Which version of the driver are you currently using (The release or devel branch one) ?
Do you have other output defined or only the ekf_euler ?

Amir-Ramezani, 2020-01-08:
Hi Erwan,

I am using the version that is installed using "apt-get install ros-kinetic(melodic)-sbg-driver" (http://wiki.ros.org/sbg_driver).

Basically, I need the angular velocities that I get by subscribing to imu_data topic using ROS, and attitude that I get by subscribing to ekf_euler topic.

I am receiving only 92 fps, even after increasing the serial port speed.

Also, the frames have delay (about milliseconds), I am not sure how much, but it reduces the performance of the whole system.

Thanks,

ErwanCL, 2020-01-09:
- So you're using the master version of the driver ? (2.0.0)
- Can I see your configuration file ?
- How do you measure your frequency ?

And for the error, someone told me about it as well, for some reasons, my cmake version didn't cahce the typo error, it is corrected now !

Amir-Ramezani, 2020-01-09:
About the version, it seems to me 1.1.7?
:~$ apt-show-versions ros-kinetic-sbg-driver
```
ros-kinetic-sbg-driver:amd64/xenial 1.1.7-0xenial-20191214-110155+0000 uptodate
ros-kinetic-sbg-driver:arm64 not installed
ros-kinetic-sbg-driver:i386 not installed
```

About the frequency:
I measure it using `rostopic hz 'topic name'`

That being said, I am testing our system using other IMUs also, by looking at the results, I can see SBG is more accurate, but data arrive slower and with a delay (I said that by looking at the behaviour of the system and my experience with other IMUs).

This is my configuration file:
```
# Configuration file for SBG Ellipse
# YAML

# Ellipse-A - Magnetic-based
# Ellipse-E - External GNSS
# Ellipse-N - GNSS-based
# Ellipse-D - Dual-antenna GNSS

uartConf:
  # Port Name
  portName: "/dev/ttyUSB0"

  # Baude rate (4800 ,9600 ,19200 ,38400 ,115200 [default],230400 ,460800 ,921600)
  baudRate: 460800

  # Port Id
  # 0 PORT_A: Main communication interface. Full duplex.
  # 1 PORT_B: Auxiliary input interface for RTCM
  # 2 PORT_C: Auxiliary communication interface. Full duplex.
  # 3 PORT_D: Auxiliary input interface
  # 4 PORT_E: Auxiliary input/output interface
  portID: 0

# Sensor Parameters
sensorParameters:
  # Initial latitude (°)
  initLat: 48.419727
  # Initial longitude (°)
  initLong: -4.472119
  # Initial altitude (above WGS84 ellipsoid) (m)
  initAlt: 100.0
  # Year at startup
  year: 2018
  # month in year at startup
  month: 03
  # day in month at startup
  day: 10

  # Montion profile ID
  # 1 GENERAL_PURPOSE Should be used as a default when other profiles do not apply
  # 2 AUTOMOTIVE      Dedicated to car applications
  # 3 MARINE          Used in marine and underwater applications
  # 4 AIRPLANE        For fixed wings aircraft
  # 5 HELICOPTER      For rotary wing aircraft
  motionProfie: 1

# IMU_ALIGNMENT_LEVER_ARM
imuAlignementLeverArm:
  # IMU X axis direction in vehicle frame
  # 0 ALIGNMENT_FORWARD   IMU Axis is turned in vehicle's forward direction
  # 1 ALIGNMENT_BACKWARD  IMU Axis is turned in vehicle's backward direction
  # 2 ALIGNMENT_LEFT      IMU Axis is turned in vehicle's left direction
  # 3 ALIGNMENT_RIGHT     IMU Axis is turned in vehicle's right direction
  # 4 ALIGNMENT_UP        IMU Axis is turned in vehicle's up direction
  # 5 ALIGNMENT_DOWN      IMU Axis is turned in vehicle's down direction
[... 183 lines cut ...]
  log_gps1_vel: 0
  # GPS positions from primary or secondary GPS receiver
  log_gps1_pos: 0
  # GPS true heading from dual antenna system
  log_gps1_hdt: 0
  # GPS 1 raw data for post processing.
  log_gps1_raw: 0
  # Provides odometer velocity
  log_odo_vel: 0
  # Event A/B/C/D Event markers sent when events are detected on a sync in pin
  log_event_a: 0
  log_event_b: 0
  log_event_c: 0
  log_event_d: 0
  # Barometric altimeter output
  log_pressure: 0

  # Node frequency (Hz) (if set to 0, the node will decide the most appropriate frequency to run)
  frequency: 200

```

Yes, I also changed it to capital V!

Thanks,

ErwanCL, 2020-01-10:
Huges changes have been made on the current repository version (2.0.0) regarding the message handling, so I recommend you to use this version (which should be available via apt-get soon).

> Just wondering how it automatically goes to 200 hz even though I am using the default uart setting file which sets the output to be 25 hz.

If you didn't reconfigure your device, even if the settings file is set to 25Hz, as your Ellipse is currently sending data at 200Hz.
A parameter is available in the config file to configure again your device with the driver.

Amir-Ramezani, 2020-01-10:
Okay, great!

Can I get data higher than 200 Hz per second?

ErwanCL, 2020-01-10:
It is not possible to configure sbg output at more than 200Hz.

closed 2020-01-20 by ErwanCL