SBG-Systems/sbg_ros_driver #27: Ekinox launch file

KevinBdn, 2020-01-17:
Hi,
Would it possible to have a launch file for an Ekinox device and is the driver compatible with this device?

Best,

ErwanCL, 2020-01-20:
Hello,

The driver is compatible with Ekinox (and Apogee as well) devices.
In case of Ethernet connection with the Ekinox, you can modify these files to get the driver working.

- in the launch file `sbg_device.launch`, change the setting file, to use the udp one `<rosparam command="load" file="$(find sbg_driver)/config/sbg_device_udp_default.yaml" />`
- in the udp settings file, just configure the IP configuration
`# Udp configuration
ipConf:
  ipAddress: "0.0.0.0" out_port: 1234  in_port: 5678`, and set your corresponding local parameters.

Be careful, the sbg commands are not available for the Ekinox (see the firmware manual), so you will have to configure your Ekinox through the web interface, and define the corresponding output topics in your setting files.

KevinBdn, 2020-01-27:
Thanks,

I managed to use the package with an Equinox, but only with a know IP address. I'm not sure if the socket configuration allows a broadcast method. Do you know if a specific configuration is necessary for the broadcast mode ?

ErwanCL, 2020-01-28:
You can configure your Ekinox through the web interface to allow the broadcast mode.
When it is done, you can use the driver in broadcast with your Ekinox, just knowing the input/output ports. (I have tested it on an Ekinox-A, all works properly)

WARNINGs. 
- The IP address is required to send commands to the Ekinox, so in broadcast, you won't be able to identify it at the start of the driver, but it will work anyway.
- As the IP address won't be defined, you can have conflict issues if you're using several products with the same input/output ports.

Best regards

closed 2020-11-09 by rsiryani