SBG-Systems/sbg_ros_driver #28: sbg_driver fails to build on Kinetic

tfoote, 2020-01-23:
It looks like a missing dependency. 

From: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__sbg_driver__ubuntu_xenial_amd64__binary/22/console

```
00:02:02.706 -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
00:02:02.713 CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
00:02:02.713   Could not find a package configuration file provided by "eigen_conversions"
00:02:02.713   with any of the following names:
00:02:02.713 
00:02:02.713     eigen_conversionsConfig.cmake
00:02:02.713     eigen_conversions-config.cmake
00:02:02.713 
00:02:02.713   Add the installation prefix of "eigen_conversions" to CMAKE_PREFIX_PATH or
00:02:02.713   set "eigen_conversions_DIR" to a directory containing one of the above
00:02:02.713   files.  If "eigen_conversions" provides a separate development package or
00:02:02.713   SDK, be sure it has been installed.
00:02:02.713 Call Stack (most recent call first):
00:02:02.713   CMakeLists.txt:4 (find_package)
00:02:02.713 
00:02:02.713 
00:02:02.713 -- Could not find the required component 'eigen_conversions'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
00:02:02.713 CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
00:02:02.713   Could not find a package configuration file provided by "eigen_conversions"
00:02:02.713   with any of the following names:
00:02:02.713 
00:02:02.713     eigen_conversionsConfig.cmake
00:02:02.713     eigen_conversions-config.cmake
00:02:02.713 
00:02:02.713   Add the installation prefix of "eigen_conversions" to CMAKE_PREFIX_PATH or
00:02:02.713   set "eigen_conversions_DIR" to a directory containing one of the above
00:02:02.713   files.  If "eigen_conversions" provides a separate development package or
00:02:02.713   SDK, be sure it has been installed.
00:02:02.713 Call Stack (most recent call first):
00:02:02.713   CMakeLists.txt:4 (find_package)
00:02:02.713 
00:02:02.713 
```

ErwanCL, 2020-01-24:
Hello,

Indeed, it was a missing dependency on eigen_conversions.
I removed eigen from the driver, so it should be good now, and I will release the new (fixed) version, today.

Thanks,
Erwan

ErwanCL, 2020-01-28:
Fix with the remove of eigen dependencies (fd41e9a4d5af8c7a0ca341dc566f8588f52c03e2)

closed 2020-01-28 by ErwanCL