SBG-Systems/sbg_ros_driver #30: Get position

scanf30, 2020-02-17:
Hi. I'm using an Ellipse2-D IMU to get the orientation and position in a boat.
I don't know if my configuration is right and how I could get the position.
here is my configuration: 

`# Configuration file for SBG device through an Uart interface.

# Configuration of the device with ROS.
confWithRos: true

# Uart configuration
uartConf:
  # Port Name
  portName: "/dev/ttyUSB1"

  # Baude rate (4800 ,9600 ,19200 ,38400 ,115200 [default],230400 ,460800 ,921600)
  baudRate: 115200

  # Port Id
  # 0 PORT_A: Main communication interface. Full duplex.
  # 1 PORT_B: Auxiliary input interface for RTCM
  # 2 PORT_C: Auxiliary communication interface. Full duplex.
  # 3 PORT_D: Auxiliary input interface
  # 4 PORT_E: Auxiliary input/output interface
  portID: 0

# Sensor Parameters
sensorParameters:
  # Initial latitude (°)
  initLat: 48.419727
  # Initial longitude (°)
  initLong: -4.472119
  # Initial altitude (above WGS84 ellipsoid) (m)
  initAlt: 100.0
  # Year at startup
  year: 2018
  # month in year at startup
  month: 03
  # day in month at startup
  day: 10

  # Montion profile ID
  # 1 GENERAL_PURPOSE Should be used as a default when other profiles do not apply
  # 2 AUTOMOTIVE      Dedicated to car applications
  # 3 MARINE          Used in marine and underwater applications
  # 4 AIRPLANE        For fixed wings aircraft
  # 5 HELICOPTER      For rotary wing aircraft
  motionProfile: 1

# IMU_ALIGNMENT_LEVER_ARM
imuAlignementLeverArm:
  # IMU X axis direction in vehicle frame
  # 0 ALIGNMENT_FORWARD   IMU Axis is turned in vehicle's forward direction
  # 1 ALIGNMENT_BACKWARD  IMU Axis is turned in vehicle's backward direction
  # 2 ALIGNMENT_LEFT      IMU Axis is turned in vehicle's left direction
  # 3 ALIGNMENT_RIGHT     IMU Axis is turned in vehicle's right direction
  # 4 ALIGNMENT_UP        IMU Axis is turned in vehicle's up direction
  # 5 ALIGNMENT_DOWN      IMU Axis is turned in vehicle's down direction
  axisDirectionX: 0
  # IMU Y axis direction in vehicle frame
  # 0 ALIGNMENT_FORWARD   IMU Axis is turned in vehicle's forward direction
  # 1 ALIGNMENT_BACKWARD  IMU Axis is turned in vehicle's backward direction
  # 2 ALIGNMENT_LEFT      IMU Axis is turned in vehicle's left direction
  # 3 ALIGNMENT_RIGHT     IMU Axis is turned in vehicle's right direction
  # 4 ALIGNMENT_UP        IMU Axis is turned in vehicle's up direction
  # 5 ALIGNMENT_DOWN      IMU Axis is turned in vehicle's down direction
  axisDirectionY: 3
  # Residual roll error after axis alignment rad
  misRoll: 0
  # Residual pitch error after axis alignment rad
  misPitch: 0
[... 165 lines cut ...]
  log_utc_time: 200
  # Magnetic data with associated accelerometer on each axis
  log_mag: 0
  # Magnetometer calibration data (raw buffer)
  log_mag_calib: 0
  # GPS velocities from primary or secondary GPS receiver
  log_gps1_vel: 0
  # GPS positions from primary or secondary GPS receiver
  log_gps1_pos: 0
  # GPS true heading from dual antenna system
  log_gps1_hdt: 0
  # GPS 1 raw data for post processing.
  log_gps1_raw: 0
  # Provides odometer velocity
  log_odo_vel: 0
  # Event A/B/C/D Event markers sent when events are detected on a sync in pin
  log_event_a: 0
  log_event_b: 0
  log_event_c: 0
  log_event_d: 0
  # Barometric altimeter output
  log_pressure: 0

  # Node frequency (Hz) (if set to 0, the node will decide the most appropriate frequency to run)
  frequency: 30`

remiburtin, 2020-02-18:
Hi.
I see in your configuration that only the orientation is output (```log_ekf_quat: 8```).
In order to get the position, you should enable the "EKF Nav" log: ```log_ekf_nav: 8``` to make the Ellipse outputs the position at 25 Hz.

scanf30, 2020-02-18:
Ohh I see. Thank you so much. 
Another thing. The IMU starts whit huge values in all axes. Do you know how I can set it to 0?

rsiryani, 2020-02-24:
Hi,

It's perfectly normal to have large quantities for standard deviations (position/velocities and so on)
If you would like to set an initial latitude/longitude you can use the command sbgEComCmdSensorSetInitCondition for instance.
But please be aware that until the unit is initialized and using the GNSS aiding the position/velocity will drift freely.

closed 2020-11-09 by rsiryani