SBG-Systems/sbg_ros_driver #31: /imu/velocity msg calculation

giwrgosg, 2020-02-18:
I am using the Ellipse 2A and since it does not provide linear velocities, I was curious about the `/imu/velocity` msg, which outputs linear velocities.

After observing the way the linear velocities are calculated below I have a question.

```
const geometry_msgs::TwistStamped MessageWrapper::createRosTwistStampedMessage(const sbg_driver::SbgImuData& ref_sbg_imu_msg, const sbg_driver::SbgImuData& ref_p_sbg_imu_msg) const
{
  geometry_msgs::TwistStamped twist_stamped_message;
  double delta_t;

  delta_t = (ref_sbg_imu_msg.time_stamp - ref_p_sbg_imu_msg.time_stamp) * 1e-6;

  twist_stamped_message.header        = createRosHeader(ref_sbg_imu_msg.time_stamp);
  twist_stamped_message.twist.angular = ref_sbg_imu_msg.gyro;

  twist_stamped_message.twist.linear.x = (ref_sbg_imu_msg.accel.x - ref_p_sbg_imu_msg.accel.x) / delta_t;
  twist_stamped_message.twist.linear.y = (ref_sbg_imu_msg.accel.y - ref_p_sbg_imu_msg.accel.y) / delta_t;
  twist_stamped_message.twist.linear.z = (ref_sbg_imu_msg.accel.z - ref_p_sbg_imu_msg.accel.z) / delta_t;

  return twist_stamped_message;
}
```

It seems that the linear velocities are calculated by considering the derivative of linear acceleration. Shouldn't we integrate the linear acceleration instead?

rsiryani, 2020-02-24:
Hi,

Thank you for reporting that issue. This code indeed seems to be incorrect. We will have a look to fix it and use the correct quantities from the unit (velocity channel directly).

Could you please confirm if you are interested in NED or body velocity ?
Just integrating the accelerations isn't a good option as you still have a lot of unaccounted errors (bias/SF) as well as the gravity.

To get the velocity you should only use the velocity output from the ELLIPSE. However, please keep in mind that only ELLIPSE-E / N/  D INS are able to compute and output a velocity. The ELLIPSE-A is only an AHRS (attitude/heading).

giwrgosg, 2020-02-24:
> Could you please confirm if you are interested in NED or body velocity ?
The body velocity.

> To get the velocity you should only use the velocity output from the ELLIPSE. However, please keep in mind that only ELLIPSE-E / N/ D INS are able to compute and output a velocity. The ELLIPSE-A is only an AHRS (attitude/heading).
Yes I know, I was just curious how the linear velocities were produced. So I guess I need to use one of the other SBG models.

closed 2020-11-09 by rsiryani