SBG-Systems/sbg_ros_driver #34: Acceleration values from the Magnetometer

aishanou-farmwise, 2020-04-07:
Hi I am using the Ellipse 2A IMU and observing that there are acceleration measurements from the magnetometer as well. They are also slightly different from the accelerometer's readings. 

What is the significance of acceleration readings from the magnetometer? Does it have its own accelerometer? I couldn't find this in the documentation.

rsiryani, 2020-04-20:
Hi,
There is only one accelerometer in the ELLIPSE.
We can see some slight difference between the standard IMU accelerometer output and the values from the magnetometer message because of filters applied on the data.
Indeed, accelerometers output in the magnetometer log are low pass filtred to avoid aliasing issues and align the phase response to the magnetometers one.
I hope it answers your question.

aishanou-farmwise, 2020-04-20:
> Hi,
> There is only one accelerometer in the ELLIPSE.
> We can see some slight difference between the standard IMU accelerometer output and the values from the magnetometer message because of filters [...]

Thank you for the response. Just to confirm, is this low pass filter applied within the firmware?

closed 2020-11-09 by rsiryani