SBG-Systems/sbg_ros_driver #41: Ellipse2 N sensor setup issue

CraigNewlands, 2021-02-11:
I am having issues setting up the Ellipse2 N sensor. I have tried both building from source and building from packages but I get a different error each time. I am running Foxy on ubuntu 20.04. 

**Building from source**
When I build from source and run the ros2 node with default configurations:

> ros2 launch sbg_driver sbg_device_launch.py

I get the following error:

> craig@craig-Inspiron-7580:~/colcan_ws$ ros2 launch sbg_driver sbg_device_launch.py
> [INFO] [launch]: All log files can be found below /home/craig/.ros/log/2021-02-03-19-20-19-009923-craig-Inspiron-7580-12166
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [sbg_device-1]: process started with pid [12171]
> [sbg_device-1] [INFO 1612380019.092217142] [sbg_device]: SBG DRIVER - Init node, load params and connect to the device. (main() at /home/craig/colcan_ws/src/sbg_ros2/src/main.cpp:12)
> [sbg_device-1] sbgInterfaceSerialCreate: tcgetattr fails.
> [sbg_device-1] [ERROR 1612380019.095567649] [sbg_device]: SBG_DRIVER - SBG_DRIVER - [Init] Unable to initialize the interface - SBG_ERROR: error not set (main() at /home/craig/colcan_ws/src/sbg_ros2/src/main.cpp:31)
> [INFO] [sbg_device-1]: process has finished cleanly [pid 12171]

**Building from Foxy package**
Building from Foxy package and running the ros2 node with default configurations:

> ros2 launch sbg_driver sbg_device_launch.py

This node runs but I get warnings as I have not yet ran the magnetic calibration node. 

> craig@craig-Inspiron-7580:~$ ros2 launch sbg_driver sbg_device_launch.py
> [INFO] [launch]: All log files can be found below /home/craig/.ros/log/2021-02-11-19-45-33-616262-craig-Inspiron-7580-8897
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [sbg_device-1]: process started with pid [8899]
> [sbg_device-1] [INFO] [1613072733.693118348] [sbg_device]: SBG DRIVER - Init node, load params and connect to the device.
> [sbg_device-1] [INFO] [1613072734.040956705] [sbg_device]: SBG_DRIVER - productCode = ELLIPSE2-N-G4A3-B2
> [sbg_device-1] [INFO] [1613072734.041263511] [sbg_device]: SBG_DRIVER - serialNumber = 45004419
> [sbg_device-1] [INFO] [1613072734.041297259] [sbg_device]: SBG_DRIVER - calibationRev = 1.1.0.0
> [sbg_device-1] [INFO] [1613072734.041317311] [sbg_device]: SBG_DRIVER - calibrationDate = 0 / 0 / 2000
> [sbg_device-1] [INFO] [1613072734.041341692] [sbg_device]: SBG_DRIVER - hardwareRev = 1.2.0.0
> [sbg_device-1] [INFO] [1613072734.041362531] [sbg_device]: SBG_DRIVER - firmwareRev = 1.7.1438-stable
> [sbg_device-1] [INFO] [1613072734.041381376] [sbg_device]: SBG DRIVER - Initialize device for receiving data
> [sbg_device-1] [WARN] [1613072734.049047195] [sbg_device]: SBG_DRIVER - [Publisher] SBG Mag data output are not configured, the standard Magnetic publisher can not be defined.
> [sbg_device-1] [WARN] [1613072734.049121932] [sbg_device]: SBG_DRIVER - [Publisher] SBG AirData output are not configured, the standard FluidPressure publisher can not be defined.
> [sbg_device-1] [INFO] [1613072734.050543651] [sbg_device]: SBG DRIVER - START RECEIVING DATA
> [sbg_device-1] [INFO] [1613072734.091140803] [Message wrapper]: A full valid UTC log has been detected, timestamp will be synchronized with the UTC data.

Running the magnetic calibration node:

> ros2 launch sbg_driver sbg_device_mag_calibration_launch.py

Produces the following error:

>  craig@craig-Inspiron-7580:~$ ros2 launch sbg_driver sbg_device_mag_calibration_launch.py
> [INFO] [launch]: All log files can be found below /home/craig/.ros/log/2021-02-11-19-50-08-506210-craig-Inspiron-7580-9089
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [sbg_device_mag-1]: process started with pid [9091]
> [ERROR] [sbg_device_mag-1]: process has died [pid 9091, exit code -11, cmd '/opt/ros/foxy/lib/sbg_driver/sbg_device_mag --ros-args --params-file /opt/ros/foxy/share/sbg_driver/config/example/ellipse_E_default.yaml'].

Any help would be much appeciated.

rsiryani, 2021-09-03:
You should probably enable the SBG_ECOM_LOG_MAG & MAG_CALIB messages to get the magnetic calibration working.

I close this issue but feel free to re-open it if you sill have issues.

closed 2021-09-03 by rsiryani

0zzy0sbourne, 2022-12-15:
Hello. How can we enable the messages that you mentioned. Because I get the same error as well.