SBG-Systems/sbg_ros_driver #43: How to get the /sbg/gps_raw with Rinex format

yulan0215, 2021-06-16:
Hi:
I used the SBG Ellipse and with this driver I got the topic: /sbg/gps_raw, however, I would like to convert the format into Rinex format, which is the standard format of the GPS format.
Could you tell me how to implement? I will appreciate if you have some ideas about it.

nachovizzo, 2021-06-17:
I'm also interested in this

AnthonySBG, 2021-08-19:
Hello,
For this, you just have to record the binary stream of the gps_raw and use a converter tool which handle the GNSS receiver format (Ellipses v3 have a u-blox one).
You can also use Qinertia to do this.

Thank you.

nuneslu, 2022-11-21:
Hi, I have been trying to parse the raw gps information and convert it to RINEX format but it haven't been working that easily. We have directly dumped the raw data from rostopic to a `.bin` file, then we tried to use RTKLIB to convert it but it could not parse it. Is there any intermediary step missing? Or any FREE converter that you would suggest?

rsiryani, 2022-11-28:
Hi,

If you have a full sbgECom binary stream, you should probably extract the ublox raw binary stream only from the SBG_ECOM_LOG_GPS1_RAW.
You can use the sbgECom library to do so and check the sbgECom basicLogger example as well.

[sbgECom](https://github.com/SBG-Systems/sbgECom)

You can also use the SbgGpsRaw.msg ROS message.

However, please keep in mind, we (SBG) don't support RTKLib and, according to our tests and several users feedbacks, RTKLib is not able to process correctly u-blox data for Kinematic trajectories.

RTKLib with ublox is working okay for static acquisitions only but is not reliable for kinematics data. Same for the converter, it produces RINEX that are not 100% compliant with the standard and it could lead to issues.

Could you please elaborate about the GNSS / INS post processing software you would like to use?
Of course, I would recommend using Qinertia that processes these data out of the box with and ELLIPSE and any ublox receiver.

[Qinertia](https://www.sbg-systems.com/products/qinertia-ins-gnss-post-processing-software/)

Qinertia can also be used through a CLI/ REST API with interesting pricing depending on your application.

If you are interested, please feel free to contact your sales rep. to get more details.

I hope it helps,

Regards,

Raphael Siryani

nuneslu, 2022-11-28:
Hi, thanks for the quick reply. We could actually convert the data using RTKLib now. But we noticed that it does output just the `.obs` files and no `.nav`. We thought it was the problem with RTKLib and therefore we tried with the [TEQC](https://www.unavco.org/software/data-processing/teqc/teqc.html) converter but it also just output the `.obs` files. Do we have to configure the sbg device to output both `.nav` and `.obs` information in the `raw_gps` topic?

laxnpander, 2023-11-01:
@nuneslu Do you perform any intermediary steps now or what was the issue originally? This could help other users in the future.

andrewcaunes, 2024-07-12:
@nuneslu Have you found a solution for the missing .nav file please ?

nuneslu, 2024-07-12:
Unfortunately, we've found no other solution. We decided to go with RTK correction which is supported by SBG directly.