SBG-Systems/sbg_ros_driver #66: Heading and pitch accuracies missing on SbgGpsHdt message

jajberni, 2021-09-21:
It looks like the heading and pitch accuracies are parsed:

https://github.com/SBG-Systems/sbg_ros_driver/blob/00b1aa50ef0cd75519038879ab0e6837fa0866fa/external/sbgECom/src/binaryLogs/sbgEComBinaryLogGps.c#L182

But not reported in the GPS heading message:

https://github.com/SBG-Systems/sbg_ros_driver/blob/00b1aa50ef0cd75519038879ab0e6837fa0866fa/src/message_wrapper.cpp#L505

The baseline field, incorporated in version 2.0 is missing in the message definition:

https://github.com/SBG-Systems/sbg_ros_driver/blob/00b1aa50ef0cd75519038879ab0e6837fa0866fa/external/sbgECom/src/binaryLogs/sbgEComBinaryLogGps.c#L187

jajberni, 2021-09-22:
We are working on it: 2d46eedc4c55dd49c4e5eb1bfce467b328af518c 

We'll be going out for testing right now. We'll let you know. 

We haven't done anything yet with the baseline length as this requires a change in the message definition and I am not sure how this will impact compatibility.

jajberni, 2021-09-22:
We have tested the code in the field and it works as expected, with correct accuracies.

The only unexpected behaviour is that the status is 64 or 65. This is odd according to the documentation or message definition. 

Regarding the baseline length. I am happy to implement and test it.

mzembsbg, 2021-09-22:
I have done the same fix as yours.
I've also added the baseline since I don't think it will cause issues. Now the code and the message definition are coherent.

mzembsbg, 2021-09-22:
The status is an error in documentation. I've updated the description.
The bit 6 is a valid flag. 0x40 (dec 64) means the baseline is valid and a valid solution has been computed.

jajberni, 2021-09-22:
Excellent! It makes sense now. :-) 

Thank you for the changes, looking forward to seeing this in the main branch!

mzembsbg, 2021-09-27:
The fix is currently only on devel branch.
This ticket should be closed on next sbg_ros_driver release.

commit b78f3f33 (2021-09-22): fix #66 - missing heading and pitch accuracies in sbgGpsHdt msg
commit 30542d88 (2021-09-22): fix #66 - status description for SbgGpsHdt msg