SBG-Systems/sbg_ros_driver #76: EllipseD not working with other USB devices

rmarcuzzi, 2021-12-15:
I'm trying to use the Ellipse D sensor using the USB cable connected to a USB hub and I'm getting SBG_TIME_OUT error, meaning that I can't connect to it.
If I connect the device directly to the USB port of the computer it works perfectly, I can run the default example. However, when I connect it to the USB hub and try to run the example, it doesn't work. I see the same behaviour when connecting another USB GPS, I get also SBG_TIME_OUT.
I've tried changing the baudrate from 115200 to 230400 and I see the same behaviour.

richardbraun, 2022-08-29:
It seems that the USB hub is the culprit then. A low quality hub could add unreasonable latency on the communication path.

richardbraun, 2022-08-29:
Also, do check that the serial port is correct after connecting through the USB hub.