SBG-Systems/sbg_ros_driver #77: How to change the frame_id of the topic “/imu/nav_sat_fix” without changing the frame_id of other topic?

goldqiu, 2022-05-08:
Hi.
How to change the frame_id of the topic “/imu/nav_sat_fix” without changing the frame_id of other topic?
                                                                                      Thank you very much!!!!

CeeSBG, 2022-05-10:
Hello in order to change the frame_id you have to use ROS package **tf** that contains different topics so it's applies a transformation on a hole node that will publish  on the **tf** topic. So in conclusion, you can't change a frame_id in only one topic. However, you can create a **subscriber node** and subscribe to one topic only and then change the frame_id for just that topic. This web page can maybe help you to understand the process [http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29#roscpp_tutorials.2FTutorials.2FWritingPublisherSubscriber.Writing_the_Subscriber_Node](url)

goldqiu, 2022-05-11:
Ok,
Thank you，I think i know what should i do.
Best regards.

closed 2022-05-31 by CeeSBG