SBG-Systems/sbg_ros_driver #78: Unable to get the device Info : SBG_TIME_OUT

MagdaZal, 2022-08-18:
Hi, When I am trying to turn on the SBG in Ubuntu 18.04 I always see this error and warnings:

[image]

But in `rostopic echo /imu/data`  the topic is perfectly working. Do you know how to solve this problem?

I use SBG Ellipse-D with /dev/ttyUSB2 (Port E). I checked the baud rate in SBG Center and .yaml file and it is on the same value 115200.
[image]

mahammadirfan, 2022-08-22:
I am having the same issue with the same device.

AnthonySBG, 2022-08-29:
Hi MagdaZal,
Your issue is totally normal. Indeed, only the serial port A is able to configure the Ellipse. In your case, as you are using the serial port E, the ROS driver is sending sbgECom messages that the device does not handle.

So you have two choices:
- Use the serial port A with ROS
- Configure the serial port E with SBG Center, and then use ROS with the serial port E with the parameter ConfWithRos: False.

Thank you.

closed 2023-07-04 by rsiryani