SBG-Systems/sbg_ros_driver #79: ROS publishing issue

CeeSBG, 2022-08-19:
A client have noticed a strange publishing behavior using the ROS driver. It appears that there is a publishing delay that occurs at a frequency of 1Hz, with the publishing occurring at a higher frequency just after to compensate. The ROS timestamps in the header seem to indicate that the data is being received and processed at the expected frequency, so it appear to just be an issue on the publishing side.

richardbraun, 2022-11-22:
Hello, and sorry for the delay. Can you please provide more details about the problem you observe, such as one of the message types affected, relevant configuration (for example, /sbg_device/output/time_reference and /sbg_device/uartConf/baudRate, and the exact data used to determine this discrepancy. Thanks.

laxnpander, 2023-02-28:
@CeeSBG Might be related: https://github.com/SBG-Systems/sbg_ros_driver/issues/83

tolesam, 2024-10-09:
It seems to be the same issue as #83 which is already solved.