SBG-Systems/sbg_ros_driver #80: Unable to get the device Info : SBG_TIME_OUT

mahammadirfan, 2022-09-06:
Hi,
I am using a SBG Ellipse-D, and having some issues with it. 
Whenever I change the IMU output data frequency from 25Hz to 200 Hz both from the sbg-system software running in my pc and in the SBG-ROS library which i have installed.
The issue is after modifying the freq., I saved it. Now I am unable to read /Initialize the SBG device in SBG-SYSTEM and through the ROS package as well.

[image]

Can anyone help me with this ?
I already checked the baudrate,port config etc.

CeeSBG, 2022-09-08:
Hello, i'm also replaying here to make it public and maybe help others. About the configuration, if you're using SBG softwares for that, it's better to disable the configuration with ROS by modifying the yaml file. 

About the data output frequency, you will need to make sure you set a higher baudrate or the port will saturate and it could explain why you don't see the data properly and why you can't configure it properly after that. We recommend a baudrate of at least 460800.

mahammadirfan, 2022-09-08:
I have changed the baud rate to 460800 and now the sbg publishes the data at 200 HZ.
Thanks!

closed 2022-09-21 by mahammadirfan