SBG-Systems/sbg_ros_driver #83: Connection delayed / not sent every second

laxnpander, 2023-02-28:
Hey,

I am observing a weird issue with data being delayed / not sent every second. See the bag I recorded below:
[image]

My ellipse N is connected via RS232 -> UART to an onboard computer. Anyone having observed something similar?

Best regards,

Alex

rsiryani, 2023-02-28:
Hi,

Thanks for reaching us.
Please make sure the serial port is not overloaded first.

[image]<https://user-images.githubusercontent.com/20966574/221801211-c209e9c3-3af6-4bcf-b76c-58b85c995d33.png>

laxnpander, 2023-03-01:
@rsiryani Now that you say it, I realise it matches the arrival of the GNSS raw data perfectly. So you are probably right! I disabled all messages that I don't need and reduced some frequencies here and there. Looks better now.

On another note: I've set my ellipse to 460800 Baud. When I go to 921600 Baud I get a timeout starting the ros node. So I assumed this is the limit for my serial connection. Do you have a recommendation on how to connect the Ellipse to an onboard computer to utilise the maximum possible bandwidth? USB and 921600 works flawlessly, but UART only on 460800. On the other hand, I need the synchronisation pins for my other sensors, so USB is not really any option.

closed 2023-03-01 by laxnpander