SBG-Systems/sbg_ros_driver #84: Synchronization of the SBG unit

yulan0215, 2023-03-30:
Hi: 

I am sorry that I have a question for the synchronization of the SBG unit ROS msg.

The is the time_ reference and synchronization instruction:
[image]

However, I did not get it, if I set the:
 
# Time reference:
time_reference: "ins_unix"

 Then all the header time from SBG unit will use UTC time? Is it correct?
However, I have alreay set like the aboved-mentioned but it seems to be that the header time is still come from ROS environment.

Do you have any idea about the synchronization? Thx

benmakka, 2023-10-17:
Hello @yulan0215, if you want to change any setting using ROS, you'll need to enable the "configWithRos" parameter in the same file to enable configuration changes through ROS. Can you confirm whether it's already set to "true" on your end?

tolesam, 2024-10-09:
Hello @yulan0215 ,
Using `time_reference: "ros"`, the timestamp will be the system's one and retrieved when sending the ROS topic.
Using `time_reference: "ins_unix"`, it will be the one sent by the INS and retrieved at the moment of the message generation in the INS. However, it requires the INS to be synced and a valid UTC message to be received by the driver which will display the following log:
`[ INFO] [1728485556.038434331]: A full valid UTC log has been detected, timestamp will be synchronized with the UTC data.`

maxhofidrone, 2024-10-29:
Hi, I have the same question like @yulan0215:
I'm using the Quanta Micro/UAV and the standard ROS Noetic driver. As @tolesam wrote this driver should show following log if `time_reference: "ins_unix"`:
`[ INFO] [1730212362.015636643]: A full valid UTC log has been detected, timestamp will be synchronized with the UTC data.`

If I set it to `time_reference: "ros"` I will get the exact same log! How can we be sure if the ROS messages contains now the correct (IMU/UTC) timestamp?

tolesam, 2024-10-31:
Hi @maxhofidrone ,
As you wrote, this log will appear regardless `time_reference`value.
If `time_reference` is set to `ros`, we'll use ROS time anyway.
If `time_reference`is set to ìns_unix`, we'll use ROS time until a first valid UTC message is received from the INS and then use it starting from this moment.

ROS messages header timestamp assignment: https://github.com/SBG-Systems/sbg_ros_driver/blob/master/src/message_wrapper.cpp#L38
Valid UTC detection log: https://github.com/SBG-Systems/sbg_ros_driver/blob/master/src/message_wrapper.cpp#L743