SBG-Systems/sbg_ros_driver #85: Changing the output frequency in config file does not change the actual output of the topic

hoangvietdo, 2023-05-16:
Hi,

I've noticed that the default output rate of the "sbg_device.launch" is ``25`` Hz.
I want to increase the output rate to ``100`` Hz, so I checked the config file "sbg_device_uart_default.yaml" and changed the parameter ``log_imu_data`` of the ``output`` section from ``8`` to ``2``.
However, when I check the topic Hz using the ``rostopic hz`` command, the rate is still ``25`` Hz.

Is it a bug?
Thank you in advance.

louis-wiesmann, 2023-10-11:
I've got the same problem. Could you solve it somehow?

hoangvietdo, 2023-10-12:
@louis-wiesmann I think the problem comes from the baud rate settings. I had to use my Windows machine to download the [sbgCenter](https://support.sbg-systems.com/sc/dev/latest/inertial-sdk) application and then change the sensor's baud rate to 921600. After that, this ros-driver can return ~100 Hz sensor data.

However, the data look very strange to me. It still looks like 25 Hz data (I need to figure out how to explain this).
So if your purpose is to record only the SBG sensor's data, then I recommend you to use the [sbgDataLogger](https://support.sbg-systems.com/sc/dev/latest/sbgdatalogger-tool) from the Linux machine and then copy the recorded file to a Window machine, use sbgCenter to convert the file to ```.txt```.

The procedure sounds very bulky, but at least it works for the mentioned purpose.

louis-wiesmann, 2023-10-13:
@hoangvietdo thank you for the response. Yeah, also for me when increasing the baudrate still results in 25Hz. Unfortunatly we rely on ros and using the sbgDataLogger is not an option.

benmakka, 2023-10-13:
The default configuration in "sbg_device_uart_default.yaml" does indeed set the output rate of the "log_imu_data" message to 25Hz. If you want to change this setting using ROS, you'll need to enable the "**configWithRos**" parameter in the same file to enable configuration changes through ROS.

Please note that while you can adjust all parameters, except for the baudrate of port A, which must be configured through the sbgCenter software. Increasing the output rate of the IMU message should generally be fine and shouldn't overload the port. However, if you decide to increase the output rate for other data, you might saturate the port. In such cases, don't hesitate to fine-tune the port A baudrate to align it with your specific use case requirements.

louis-wiesmann, 2023-10-16:
@benmakka The `configWithRos` parameter was already set in my case, but still couldn't get the desired framerate. Lowering the framerate worked, but when increasing it capped at 25Hz.

benmakka, 2023-10-16:
@louis-wiesmann Can you please confirm if you are having the issue with the same message (log_imu_data)? Could you share a rosbag file and your configuration file for reference? This will allow me to ensure everything is set up correctly.

louis-wiesmann, 2023-10-16:
Yes with `log_imu_data`. Here is the bagfile and the config we use:
[sbg_imu_hz_test.zip](https://github.com/SBG-Systems/sbg_ros_driver/files/12913794/sbg_imu_hz_test.zip)

Thank you very much!

louis-wiesmann, 2023-10-16:
Yes, this was the problem. When subscribing to `/sbg/imu_data` I receive the 100 Hz, while `/imu/data` still is 25 Hz. 

Thank you very much for the reply!

benmakka, 2023-10-16:
@hoangvietdo Could you please close the ticket ? Thank you in advance.

closed 2023-10-16 by hoangvietdo