SBG-Systems/sbg_ros_driver #86: Ellipse-D config aidingAssignment parameters

ahmetsaglam, 2023-06-07:
I am confused about **aidingAssignment** configuration in the example config file of ellipse_D_default.yaml .

The below is the part from the file:

```
# AIDING_ASSIGNMENT
# Note: GNSS1 module configuration can only be set to an external port on Ellipse-E version.
# Ellipse-N users must set this module to MODULE_INTERNAL. On the other hand, rtcmModule is only
# available for Ellipse-N users. This module must be set to MODULE_DISABLED for other users.
aidingAssignment:
  # GNSS module port assignment:
  # 255 Module is disabled
  # 1 Module connected on PORT_B
  # 2 Module connected on PORT_C
  # 3 Module connected on PORT_D
  # 5 Module is connected internally
  gnss1ModulePortAssignment: 5
  # GNSS module sync assignment:
  # 0 Module is disabled
  # 1 Synchronization is done using SYNC_IN_A pin
  # 2 Synchronization is done using SYNC_IN_B pin
  # 3 Synchronization is done using SYNC_IN_C pin
  # 4 Synchronization is done using SYNC_IN_D pin
  # 5 Synchronization is internal
  # 6 Synchronization is done using SYNC_OUT_A pin
  # 7 Synchronization is done using SYNC_OUT_B pin
  gnss1ModuleSyncAssignment: 5
  # RTCM input port assignment for Ellipse-N DGPS (see gnss1ModulePortAssignment for values)
  rtcmPortAssignment: 255
  # Odometer module pin assignment
  # 0 Odometer is disabled
  # 1 Odometer connected only to ODO_A (unidirectional).
  # 2 Odometer connected to both ODO_A (signal A) and ODO_B (Signal B or direction) for bidirectional odometer.
  odometerPinAssignment: 0
```

The comments say that  `Ellipse-N users must set this module to MODULE_INTERNAL` which is option 5.
Then, here  `gnss1ModulePortAssignment: 5` but this config is for Ellipse-D, not Ellipse-N. 
If this is correct, which I believe, it should be because Ellipse-D has internal GNSS receiver, then this statement `Ellipse-N users must set this module to MODULE_INTERNAL` should be updated to `Ellipse-D and Ellipse-N users must set this module to MODULE_INTERNAL`.

rsiryani, 2023-06-07:
Hi,

You are absolutely right.
If you use an INS with an internal GNSS receiver, you should set both port and sync to internal (ie 5).

Also if you are using an ELLIPSE-D v3 or up to date firmware for ELLIPSE-N/D, you should set the GNSS configuration model to 101 (ie internal) or keep the default settings.

We have simplified this configuration at lot as you can find in the latest sbgECom library:
- https://github.com/SBG-Systems/sbgECom/blob/main/doc/migrations.md
- https://github.com/SBG-Systems/sbgECom/blob/main/src/commands/sbgEComCmdGnss.h

Hope it helps,

closed 2023-06-07 by rsiryani

ahmetsaglam, 2023-06-07:
Great, thank you!