SBG-Systems/sbg_ros_driver #87: Ellipse-D dual antenna GNSS configuration for ROS Lunar

ahmetsaglam, 2023-06-07:
There is no example config file for Ellipse-D in ROS Lunar version. Exiting example configs do not have parameters for second antenna for GNSS configuration. Could you please provide an example for this?

```
# GNSS configuration
# Note: Pitch and Yaw offsets as well as antenna distance parameters should only be considered in
# case of dual antenna GNSS receiver. It can be left to 0 otherwise.
gnss:
  # Gnss Model Id
  # 101 Used on Ellipse-N to setup the internal GNSS in GPS+GLONASS
  # 102 Default mode for Ellipse-E connection to external GNSS
  # 103 Used on Ellipse-N to setup the internal GNSS in GPS+BEIDOU
  # 104 Used on Ellipse-E to setup a connection to ublox in read only mode.
  # 106 Used on Ellipse-E to setup a connection to Novatel receiver in read only mode.
  # 107 Used on Ellipse-D by default
  gnss_model_id: 102

  #GNSS antenna lever arm in IMU X axis (m)
  leverArmX: 0
  #GNSS antenna lever arm in IMU Y axis (m)
  leverArmY: 0
  #GNSS antenna lever arm in IMU Z axis (m)
  leverArmZ: 0
  #Pitch offset for dual antenna GNSS (rad)
  pitchOffset: 0
  #Yaw offset for dual antenna GNSS (rad)
  yawOffset: 0
  #Distance between two GNSS antennas (m)
  antennaDistance: 0

  # Rejection mode for position
  # 0 Measurement is not taken into account
  # 1 Measurement is rejected if inconsistent with current estimate (depending on error model)
  # 2 Measurement is always accepted
  posRejectMode: 1
  # Rejection mode for velocity (see posRejectMode values)
  velRejectMode: 1
  # Rejection mode for true heading (see posRejectMode values)
  hdtRejectMode: 1
```

rsiryani, 2023-06-07:
Hi,

You can find this information here:
https://github.com/SBG-Systems/sbg_ros_driver/blob/4684439988aa1268093a99efb3bfdae0caeba11b/src/config_store.cpp#LL115C11-L115C11

And here:
https://github.com/SBG-Systems/sbg_ros_driver/blob/master/config/example/ellipse_D_default.yaml

I don't know where your yaml conf is coming from but it is deprecated as know you can input primary/secondary lever arms instead of primary/yaw/pitch offsets

Please let me know if you have other questions.

ahmetsaglam, 2023-06-07:
You are that it is deprecated. I am trying to update the firmware. The current ROS package depends on `SBG communication protocol sbgECom, v1.11.920-stable (full compatible with firmwares from 1.7.x)` but we have firmware `1.3.97-stable`. 
Currently, I am trying to figure out compatibility issues, if any, within a Ubuntu 16.04, ROS Lunar environment if I update to a newer firmware. That's why I am using an older ROS package, where there are parameters for primary/secondary lever arms.

benmakka, 2023-10-17:
Hello @ahmetsaglam, were you able to download the latest ROS package version ? Are you still having difficulties with the dual antenna lever arms configuration ?

closed 2024-10-09 by tolesam