SBG-Systems/sbg_ros_driver #93: Node dies at startup with `Unable to get output for the device : Class [0] - Id [3] : SBG_TIME_OUT`

PerrineAguiar, 2024-01-24:
Hi
We are using the sbg ros driver (ROS1) with `confWithRos` set to true, and we often encounters the following issue:
the node dies at startup with the error message
```
[ERROR] [1705600023.845223066]: SBG_DRIVER - [Config] Unable to get output for the device : Class [0] - Id [3] : SBG_TIME_OUT
```
the `Id` in the message may vary, but the error is always `SBG_TIME_OUT`.
The issue does not always happen, but quite often (~50% of the time). For now our only solution is to restart the node until it works.

Looking a bit into the code, I tried increasing [numTrials](https://github.com/SBG-Systems/sbg_ros_driver/blob/4684439988aa1268093a99efb3bfdae0caeba11b/external/sbgECom/src/sbgECom.c#L37) or [cmdDefaultTimeOut](https://github.com/SBG-Systems/sbg_ros_driver/blob/4684439988aa1268093a99efb3bfdae0caeba11b/external/sbgECom/src/sbgECom.c#L38) but it did not help.

The only workaround to solve this issue seems to be that we must set `confWithRos` to false and make sure our SBG Ellipse-D is configured correctly in the sbgCenter Windows tool. But this is not ideal so I was wondering if this is a known issue and if there is a proper way to fix it.

tolesam, 2024-10-09:
Hello @PerrineAguiar ,
Thank you for your feedback.
I can't reproduce your issue, can you try again with the latest sbg_ros_driver version?
We did several improvements on it, it could solve your issue.
Moreover can provide your settings and driver's logs please?

PerrineAguiar, 2024-10-28:
Hi @tolesam 
In the meantime we switched to the [SBG ROS2 driver](https://github.com/SBG-Systems/sbg_ros2_driver) and did not encounter this issue anymore.

tolesam, 2024-10-28:
Hi @PerrineAguiar 
Thank you for your answer, that's a good news.
I close the issue then.

closed 2024-10-28 by tolesam